PhotoboothProject/photobooth #849: WLED API support

phn2211, 2024-09-08:
### Is your feature request related to a problem?

Yes

### Description

Hi, a neopixel LED ring added to the construction ist either an eyecatcher and makes concentration in the lens If in countdown the LEDs turn off one by one.

The WLED Project could be included i think. An inputfield for the IP address and two inputfields for the effect for 
countdown in ring with a post request in hitting "take picture" and one for Basic animation would be really great

### Describe the solution you'd like

3 Input fields for configuration.
Do an API Post Request in taking Photo to Run Animation Countdown. After Photo was made send another request for Basic Animation Like rainbow

### Describe alternatives you've considered

Look before

andi34, 2024-09-09:
You should be able to use the existing get request function.

[image]

Only after processing we might need to add a new option as it's only possible to send the photo style ATM.

Following the first example from https://kno.wled.ge/interfaces/http-api/ would look like this:
[image]

andi34, 2024-09-09:
You could also use the pre and post photo commands
[image]

closed 2025-02-23 by andi34