PhotoboothProject/photobooth #871: [BUG]: very long shtter lag after cheeeeese

hubertmeier, 2024-09-22:
### Describe the bug

i installed the script after a fresh bookworm installation on a raspberry pi 4. The webui works fine, the load is always under 5%, the pi has nothing to do.
If i take a picture (gphoto2 with Sony Alpha 7C or another Sony Alpha) it requires 15-20 seconds to the shutter! If i tke the picture from the terminal with the gphoto2 command, it takes 2-3 seconds.
I see in htop, that after 3-2-1-cheeese 10-15 seconds pass away before the job of gphoto2 appears! The load of the pi is always under 5%.
After gphoto2, the animation with the circle appears, processing of the picture requires 20-30% load.
What can i do that photobooth starts gphoto2 immediately?

Thank you, its a nice software.

### Steps to reproduce the behavior

fresh raspberry os on rpi 4
Sony Alpha 7c

### Expected behavior

taking a picture immediately or starting gphoto2 immediately

### Photobooth Version

4.5.1

### Browser and browser version

firefox on windows, firefox on android

### Operating System

raspberry os debian 12

### Did it work before?

No

andi34, 2024-09-22:
You can try one of the alternative capture commands mentioned in the FAQ.
Please also take a look at the browser console while Loglevel is set to 2.

https://photoboothproject.github.io/faq/#alternative-capture-commands

Photobooth v4 stable branch only gets compatibility updates, bugs won't be looked at.

If the issue persist please install latest development version and check there.

mhellmeier, 2024-12-25:
I am currently having similar issues with a fresh installation. I tried to debug the problem with a system of a 5-second countdown with a live view of the DSLR camera.

After clicking on the take picture button, the countdown starts, and the live view is shown immediately in the background (which is great). After 5 seconds, "Cheeeeeeese" is displayed on the screen, and the Console log shows the `Capture image.` message. Just afterward, the camera makes the picture with the flash (also directly after the countdown ends, without a long waiting time). Until now, everything works as expected. After this, the photobooth stays in the "Cheeeeeeese" screen, even if the picture is taken. After a waiting time (which is the problem and too long), the console says `Taking picture took 12860ms`. The rest of the process is as expected. The processing screen is shown for a short time and the image is correctly shown on the screen. So the problem must be related while "Cheeeeeese" is shown, right after the `Capture image.` message. The 12860ms for taking a picture is way too long.

https://github.com/PhotoboothProject/photobooth/blob/fe24e40ed73715e122eef29171b4a50f54ff66f5/assets/js/core.js#L575

Maybe it is also a configuration problem. Any idea why it takes so long?

andi34, 2024-12-25:
@mhellmeier like mentioned, try one of the mentioned commands. If using go2rtc for preview simply replace the word `gphoto2` with the word `capture`.
The JavaScript is waiting for the response of the capture command to continue and end the cheese.

andi34, 2024-12-25:
And also use a cheese time of 0 if using live preview

closed 2024-12-25 by andi34