PhotoboothProject/photobooth #877: Printing isnt working

thetechnikfreak, 2024-09-25:
### Describe the bug

I cant Print the printer is connected and select as Standart printer using LP when i Print with Raspberry ui its working but Not via the cmd it Starts the Connection but transmits Zero Bytes after Testing around i found Out that the only Thing that inst working is printing jpgs via the cmd i dont know why

### Steps to reproduce the behavior

Print

### Expected behavior

Print

### Photobooth Version

4.99

### Browser and browser version

Chrome

### Operating System

Raspberry, Windows,Androt

### Did it work before?

No

andi34, 2024-09-26:
Make sure printer is shared. 
Maybe run `sudo cupsctl --share-printers` via terminal.

On Canon selphy printer it's important to use latest firmware.

Another issue might be Gutenprint or cups which might need an update.

All of possible issues are not related to Photobooth itself and OS / OS Setup related.

closed 2024-09-26 by andi34

andi34, 2024-09-26:
In case firmware is up to date and sharing printer via cmd or via cups doesn't solve your problem:

https://www.peachyphotos.com/blog/stories/building-modern-gutenprint

If updating Gutenprint also does not work, compile cups yourself from latest **2.4x branch (!)**

https://github.com/OpenPrinting/cups/blob/2.4.x/INSTALL.md

thetechnikfreak, 2024-09-26:
Hat alles nicht geholfen habe es aber jetzt so gelöst (Drucker zum testen Canon ts8351)

erst mit ```
sudo usermode -aG sudo www-data
``` Der Nutzergruppe der Fotobox Root geben

Dann ```
lpoptions -d druckername
```
Als Befehl im Admin Menü fürs drucken angeben einmal drucken (es geht nicht wenn man es einfach nur in die cmd eingibt)

(Druckername kann mit ```
lpstat -d 
``` in der cmd herausgefunden werden)

Und zuletzt folgenden command```
convert %s tempfile.png && lp -o landscape tempfile.png && rm tempfile.png
``` im Admin Menü als Befehl fürs drucken einstellen dann ging es bei mir