PhotoboothProject/photobooth #878: [BUG]: PHP imageflip reversed

theamnesic, 2024-09-26:
### Describe the bug

When using the Preview flip option, the flip is reversed in the collage.

### Steps to reproduce the behavior

Set Preview flip to horizontal or vertical and take a collage.
The rotation is wrong.

### Expected behavior

The flip is the same in the preview and on the collage.

The problem is that [PHP imageflip](https://www.php.net/manual/en/function.imageflip.php) use `IMG_FLIP_HORIZONTAL` for our `flip-vertical` and `IMG_FLIP_VERTICAL` for 'flip-horizontal' directions.

This fix the problem, PhotoboothCapture.php line 77 :

```php
case 'flip-horizontal':
    imageflip($im, IMG_FLIP_VERTICAL);
    break;
case 'flip-vertical':
    imageflip($im, IMG_FLIP_HORIZONTAL);
    break;
```

### Photobooth Version

v4.5.1

### Browser and browser version

Chrome up to date

### Operating System

Ubuntu 24.04

### Did it work before?

Yes

andi34, 2024-09-28:
Hey and thanks for your report!
PhotoboothCapture.php does not exist on Photobooth v4.5.1.

The related code seems fine to me:
https://github.com/PhotoboothProject/photobooth/blob/a8841409ed8d0db485191269d6c302505b67f64f/api/takePic.php#L23-L34

https://github.com/PhotoboothProject/photobooth/blob/a8841409ed8d0db485191269d6c302505b67f64f/lib/applyEffects.php#L27-L33

https://github.com/PhotoboothProject/photobooth/blob/a8841409ed8d0db485191269d6c302505b67f64f/lib/collage.php#L71-L78

On latest development version the related code also looks fine:
https://github.com/PhotoboothProject/photobooth/blob/90b0696e3af3bf899c279736f006f68388e7d8c7/src/PhotoboothCapture.php#L76-L88

https://github.com/PhotoboothProject/photobooth/blob/90b0696e3af3bf899c279736f006f68388e7d8c7/src/Collage.php#L86-L95

https://github.com/PhotoboothProject/photobooth/blob/90b0696e3af3bf899c279736f006f68388e7d8c7/api/applyEffects.php#L107-L113

andi34, 2024-09-28:
Issue found, the css has been wrong and is adjusted now.

closed 2024-09-28 by andi34