PhotoboothProject/photobooth #911: Webserver Not starting

thetechnikfreak, 2024-10-26:
### Describe the bug

I started my photobooth normaly after a couple Off weeks for an Event ans the Webserver wasnt starting Up. (The booth was in drawer in storage and Off and ITS Always Not Connect to the Internet also If its running) 
Before i Put it away it worked Just fine.

When i manuly start IT via ssh IT said
```
Sat Oct 26 15:26:55.273900 2024] [core:warn] [pid 2956] AH00111: Config variable ${APACHE_RUN_DIR} is not defined
apache2: Syntax error on line 80 of /etc/apache2/apache2.conf: DefaultRuntimeDir must be a valid directory, absolute or relative to ServerRoot 
```
I googled the issue and someone Said to run `apache2ctl start`
So i did IT when i started IT now IT said :
`AH00534: apache2: Configuration error: No MPM loaded.`

### Steps to reproduce the behavior

Start Up the pi

### Expected behavior

Working Photobooth 

### Photobooth Version

4.99

### Browser and browser version

None

### Operating System

Raspi 4

### Did it work before?

Yes

### Latest working version

4.99

andi34, 2024-10-26:
Apache configuration isn't touched by Photobooth, must be an issue with your OS.
As the log said, you should check line 80 of `/etc/apache2/apache2.conf`

closed 2024-10-26 by andi34

andi34, 2024-10-26:
> How do I stop and restart the Apache server on Linux?
< 
< Execute `sudo systemctl stop apache2` to halt and `sudo systemctl start apache2` to restart the Apache server on Linux.
< Confirm status with `sudo service apache2 status`.