PhotoboothProject/photobooth #912: [BUG]: First print job occurs an error

cihat05, 2024-10-26:
### Describe the bug

After capturing a photo with the photobooth, an error occurs when attempting to print the image on the first try. Upon pressing "Print" again, the photo successfully prints.

### Steps to reproduce the behavior

Capture a photo using the photobooth.
Attempt to print the photo immediately after it displays.
Observe that an error occurs on the first print attempt.
Press "Print" again.
Notice that the photo prints successfully on the second attempt.

### Expected behavior

The photo should print successfully on the first attempt after being displayed.

### Photobooth Version

4.5.1

### Browser and browser version

Edge

### Operating System

Windows 10

### Did it work before?

No

cihat05, 2024-10-26:
a quick follow-up: 
Collages do not experience this error and print successfully on the first attempt.

andi34, 2024-10-27:
Please set Loglevel 2 via Adminpanel first. Without logs nothing can be done to find out what's the root of your bug.

After another failed print, please check the Photobooth log via Debugpanel and post or attach it here.

Since we're only using the OS command line the issue could also be related to something else independent of Photobooth.

Please also note, that the source of v4.5.1 isn't up to date. We're working on a Photobooth v5, but there's no ETA due to private life.
If you're interested in latest development version please Join our telegram community.

cihat05, 2024-10-28:
Hey, thank you for your reply. I just set the loglevel via 2. 
Strangely, the failed print job is not logged. The second attempt, which is printed successfully, is logged. Therefore, I cannot upload an error log here, as it appears that everything is functioning properly.

andi34, 2024-10-28:
Well, without logs there's nothing to debug and closing the issue here. I've recently uploaded an up to date version in our telegram community, feel free too test with that.

closed 2024-10-28 by andi34

jpara, 2024-11-14:
I have exactely the same issue as @cihat05 
Is there any possibility we can support with data to analyze? Sadly, the log is empty, it only shows the second attempt