PhotoboothProject/photobooth #942: [BUG]: Live Preview onyl works on first colage photo

srvoets, 2024-11-01:
### Describe the bug

Live preview loads for first collage photo, then during countdown for second photo, the "preview" of the first image remains.

### Steps to reproduce the behavior

Take collage photo with live preview

### Expected behavior

I should get a live preview before each photo during countdown

### Photobooth Version

4.99.0

### Did it work before?

Yes

### Latest working version

I dont recall, it was in august when I tested last.

lattaai2, 2024-11-01:
Hello, I tried to fix the issue

This is what I did:
I've identified the issue in the `core.js` file. The problem occurs because the preview is not being reset between collage photos. To fix this, I've modified the `api.thrill` function to reset the preview before each collage photo is taken. Here are the changes I made:

I also created pull request: https://github.com/PhotoboothProject/photobooth/pull/944

> [!CAUTION]  
> Disclaimer: This fix was created by Latta AI and you should never merge before you check the correctness of generated code!
> 
> The fix provided by Latta AI might not be complete and it can serve as an inspiration.

---
This bug was fixed for free by Latta AI - https://latta.ai/ourmission

If you no longer want Latta AI to attempt fixing issues on your repository, you can block this account.

reloxx13, 2025-11-09:
I guess easiest way to fix would be to use the addCacheBustingParam on the background-image.

nbosb, 2025-11-22:
I'm using the same configuration (go2rtc 1.9.7 + MJPEG stream) and the same URL as you (just with a different IP). I'm testing on an iPad 6th Gen running iOS 17.7.10. I tried your suggested fix of switching from an `<img>` element back to a `<div>`. I've committed the changes to a branch: [https://github.com/nbosb/photobooth/tree/dev-preview](https://github.com/nbosb/photobooth/tree/dev-preview).

However, with this change, I’m encountering an issue where the stream is still being fetched even after the preview is hidden. As a result, the camera isn’t being released by go2rtc, and the picture-taking process fails.

reloxx13, 2025-11-22:
> My fault, my capture command was broken. I could fix my issue by using the `/usr/local/bin/capture` script that stops go2rtc.
> 
> The `<div>` element works for me, too.

heya,
nono, im having the same issues and i used the capture command. check my comments here:
https://github.com/PhotoboothProject/photobooth/commit/b25baa29ee714a9da02011e6e9b976291a897c7b#r171135714
Im on it the 2nd day not trying to find a working solution and just found this timing issue.
do you have this stop preview time before capture set in config?

nbosb, 2025-11-22:
No, I have set the _Time before the countdown ends to execute the preview stop command_ to zero. I'm using the capture script that adds a 0.5s delay between stopping go2rtc and taking the picture as described here: https://photoboothproject.github.io/faq/go2rtc-troubleshooting/#delay-capture

reloxx13, 2025-11-22:
and i dont have a cheese time which is also important for the timings: countdown, (if stop time: stop at time) countdown done, stop preview, show cheese, take pic, rt2rtc stop, sleep 0,5, capture (optional wait event 300ms), start rt2rtc.

without cheese, stop time and sleep in alternative capture, there is no pause between preview and capture, blocking the cam

andi34, 2025-11-22:
https://photoboothproject.github.io/faq/go2rtc-troubleshooting/

andi34, 2025-11-23:
Another URL to try:
```
http://192.168.4.1:1984/stream.mjpeg?src=photobooth
```

You could also play with the go2rtc config further (/etc/go2rtc.yaml)
```
api:
  mjpeg:
    write_buffer: 0
    read_buffer: 0
```

@nbosb from what I've read MJPEG via `<img>` or `<video>` will fail on Safari and maybe other mobile Browser on Android too. We could consider adding some JavaScript and use webrtc, but currently I've not really time to look at it further.

reloxx13, 2025-11-26:
so the img tag does not work on my ipad, i dont know why it is working on yours. are you sure you have the latest code on your hosting device?

i changed it back and added cacheBusting to the div background-image and this is working.

But the latest go2rtc version 1.9.12 breaks the preview. the cam is stucking in busy or complete dark lcd when switching from video to capture.
I tried everything from the troubleshooting, trying around with the timings (up to 3seconfs sleep or wait event time) and spent alot of time getting the cache busting working and in the end it was go2rtc. 

im back on 1.9.7 and finally the cam and preview is working again. im trying to find the exact breaking version now.