PhotoboothProject/photobooth #948: [BUG]: OpaqueResponseBlocking for live preview from urls after clean install

mariusberlinghof, 2024-11-02:
### Describe the bug

I just did a clean install with the installation script on a raspi5 with a camera module V3.
After setting photobooth up (everything works), i tried setting up live preview from url with the setup script listet [here](https://photoboothproject.github.io/faq/).
[image]

Livestreams work as long as i open them directly via the browser (url: http://10.0.0.226:1984/stream.html?src=photobooth), from remote devices and the same device.
If i try taking a photo via photobooth the following message gets displayed in the console: 
[image]
Seems like something is making the browser block the request...

### Steps to reproduce the behavior

1. Install photobooth
2. install go2rtc-preview
3. set up livestream from url as preview

### Expected behavior

livestreams are shown correctly in photobooth during live preview

### Photobooth Version

4.99.0

### Browser and browser version

firefox 132.0

### Operating System

raspberry pi os with desktop

### Did it work before?

Yes

### Additional context

had the preview up and running before, cant figure out whats wrong right now other than OpaqueResponseBlocking

andi34, 2024-11-02:
Try with Chromium please, maybe downgrade Firefox.

You can also try the following:
> Try to set these prefs to false in `about:config`.
```
> browser.opaqueResponseBlocking => false
> browser.opaqueResponseBlocking.javascriptValidator => false

> You can open about:config via the location/address bar.
> On the warning page, click "Accept the Risk and Continue" to open about:config.

mariusberlinghof, 2024-11-02:
tried with chrome (windows), chromium and firefox on raspi as well, still not working.
Changing the settings didn't seem to solve the problem (although the warning disappeared)

andi34, 2024-11-02:
Can you please downgrade go2rtc to v1.9.2

```
wget -O install-go2rtc-preview.sh https://raw.githubusercontent.com/PhotoboothProject/photobooth/dev/scripts/install-go2rtc-preview.sh
sudo bash install-go2rtc-preview.sh
```

mariusberlinghof, 2024-11-02:
Already tried 1.9.2 and 1.9.6...

andi34, 2024-11-02:
Unable to reproduce. Having it working on a Pi5 and Pi Cam v3. Getting a preview using go2rtc v1.9.2, v1.9.4 and v1.9.6.
Tested remotely from a Pixel 8 (currently no display connected to the Pi)

mariusberlinghof, 2024-11-04:
will try a clean reinstall tonight. Just to be sure:

- operating system: latest from raspberry-imager (with desktop)
- installations script for photobooth as stated in documentation
- installation for preview with script as described in documentation
Which camera mode should i choose (libcamera or raspicamera)?
Will get back in touch after that...

andi34, 2024-11-04:
Rpicam-apps, choose go2rtc v1.9.2

mariusberlinghof, 2024-11-06:
just did a full clean reinstall. Still not working...

1. flushed Raspi os with desktop 64Bit
2. installed photobooth
3. installed go2rtc v1.9.2 (rpicam-apps)
4. reboot

andi34, 2024-11-06:
IP is unchanged? I'll send you my personal configuration adjusted with your needed IP next days

mariusberlinghof, 2024-11-06:
I just did some digging into go2rtc. The Problem is the config of the stream which is set up by the script.
codec should be set to 'mjpeg' (photobooth2) and not as shown in following screenshot (photobooth):
[image]

andi34, 2024-11-06:
https://github.com/PhotoboothProject/photobooth/commit/a43cba121f81f1342d04517900eb1667efb9e9cb

Should be set right on go2rtc v1.9.2.

closed 2024-11-06 by andi34

mariusberlinghof, 2024-12-05:
seems to be alright by now. Thanks for your awesome help!