PhotoboothProject/photobooth #986: [FEATURE]: Change Text-Color on User Interface

Fbn351, 2024-12-01:
### Is your feature request related to a problem?

Yes

### Description

At the Adminpanel you can change the Color for the "Countdown" and the "Cheese". Actually you cant change the Color and the bold of the follgowing Textphrases: "Start Screen", "Start Screen (Subtitle)", "Event Text Left", "Event Text Right" and "Hidden Trigger".

It is very helpful to change the Font, Font-Color and Bold/Italic of the single Phrases to have more Options to match with the background. F.e. i have for the Christmas Event a Background with Darkgreen or Brown Color. A lighter Color was perfect to match with the Background. And i other Scenes you can match the Font to the Single Event f.e. a Wedding

[image]

### Describe the solution you'd like

Because there are many texts and customization options, I have two ideas:
First: An extra menu item for Customation of Text
or
Second: Little Option Gears after the Textfield
[image]

The First one I think is easier to implement. But i dunno - i´m just an User ;)

### Describe alternatives you've considered

I dont know if there are alternatives. Because this is only a feature request.

### Additional context

Here an idea how it can look like
[image]
---
[image]
---
For this Idea i´ve copied the functional div into a new order
frontpage:event_textLeft => Textfeld
frontpage:ui_skip_welcome => On/Off Button
frontpage:ui_admin_shortcut =>  On/Off Button
pictures:textonpicture_font_color => Color
pictures:textonpicture_font => Font picker

When you make one Grid Element with two checkboxes (for me Bold and italic have to get together) you have the original 4 Column Grid.

Or you do it with a special css only for the Text Customizing with a 5 column Grid (i think its harder to implement und crash the flexibility of the Site.
[image]

Fbn351, 2024-12-01:
### Structur:
- [ ] Add Menu Item (f.e. Custom Text)
- [ ] Add Adminsection
- [ ] Add Gridsystem to the new Adminsection
- [ ] Move existing Flex Objects to the new Gridsystem:
   - [ ] Textfield Maintitle
   - [ ] Textfield Subtitle
   - [ ] Textfield Trigger Button
   - [ ] Event Text Left
   - [ ] Event Text Right
- [ ] Add new Flex Objects in Gridsystem (per Text-Object):
   - [ ] Text Option Bold & Italic (summarized in one Flex-Object)
   - [ ] Text Color
   - [ ] Font
---
### Translations:
- [ ] Add Translation for Flex Objects:
   - [ ] Textfield
   - [ ] Text Option Bold & Italic
   - [ ]  Text Option Bold & Italic Tooltip
---
### Functionallity/Changes:

- [ ] Implement Variables to the Sourcecode of the Frontpage:
   - [ ] Textfield
   - [ ] Text Option Bold & Italic
   - [ ] Text Color
   - [ ] Font

Hope i don´t forget a Step. This is also to work on the pull request.

Fbn351, 2024-12-02:
I´ve build the new Adminsection in the 
> \lib\configsetup.inc.php

for that i moved the existing ones to the new Section and add some new Parameters to the Section for the new Options.
Foreach of
- Maintitle
- Subtitle
- Event Text Left
- Event Text Right
- Buzzer Text
 
i´ve added the Option to change the
- Fontweight
- Italic
- Fontcolor
- Fontfamily

For the moved Parameters i had to change some entrys in the
> \template\components\stage.start.php
> \template\components\actionBtn.php`

I als had to change the Defaults in the
> \src\Configuration\PhotoboothConfiguration.php

and add some new Entrys for the new Options

Now i have to check if everything is perfectly written to the Personal Config
> \config\my.config.inc.php

And change the Translations in the 
> \resources\lang\*.json

In this Files i have to move the existing 7 Entries for the Objects:
`"frontpage:button_buzzer_message": "###",`
`"frontpage:event_textLeft": "###",`
`"frontpage:event_textRight": "###",`
`"general:start_screen_subtitle": "###",`
`"general:start_screen_subtitle_visible": "###",`
`"general:start_screen_title": "###",`
`"general:start_screen_title_visible": "###",`

and the 7 Entries for the Tooltips:
`"manual:frontpage:button_buzzer_message": "###",`
`"manual:frontpage:event_textLeft": "###",`
`"manual:frontpage:event_textRight": "###",`
`"manual:general:start_screen_subtitle": "###",`
`"manual:general:start_screen_subtitle_visible": "###",`
`"manual:general:start_screen_title": "###",`
`"manual:general:start_screen_title_visible": "###",`

Also i have to add the new Objects and Tooltips for
`"custom_text:start_screen_subtitle_bold": "###",`
`"custom_text:start_screen_subtitle_italic": "###",`
`"custom_text:start_screen_subtitle_color": "###",`
`"custom_text:start_screen_subtitle_font": "###",`
`"manual:custom_text:start_screen_subtitle_bold": "###",`
`"manual:custom_text:start_screen_subtitle_italic": "###",`
`"manual:custom_text:start_screen_subtitle_color": "###",`
`"manual:ustom_text:start_screen_subtitle_font": "###",`

---

After that everything is done for the Structur Part. Enough for today. GN8

reloxx13, 2025-11-28:
hey,
upload your code to a branch and i could see if i can finish this.