PhotoboothProject/photobooth #1001: [BUG]:

florianibach, 2024-12-04:
### Describe the bug

When having the Fullscreen mode toggled (via button), taking a picture / collage and being in "summary view" (where you can print and edit a picture) and then clicking on Home button on the top left corner - then the fullscreen mode is being closed again. 

When navigating back from the gallery to the main page (by clicking close in the top right corner) it is working

I am using the dev branch version with commit: 623b1777

I tested this on many different browsers and end devices (Edge, Edge mobile, Chrome, Chrome mobile, firefox mobile)

### Steps to reproduce the behavior

1. go to admin panel
2. navigate to "User interface" section
3. Toggle button "Show button to toggle fullscreenIf" so it is checked (true)
4. Navigate to main page
5. Toggle fullscreen
6. take a picture / take a collage
7. (wait till being in overview where you can see the taken picture)
8. click on "Home" in the top left corner
9. see: fullscreen is exited

### Expected behavior

The fullscreen mode should only be untoggled by the toggle button or exiting it manually via the browser

### Photobooth Version

I am using the dev branch version with commit: 623b1777

### Browser and browser version

I tested this on many different browsers and end devices (Edge, Edge mobile, Chrome, Chrome mobile, firefox mobile)

### Operating System

docker compose on windows

### Did it work before?

No

andi34, 2024-12-04:
Duplicate of https://github.com/PhotoboothProject/photobooth/issues/831

andi34, 2024-12-04:
https://developer.mozilla.org/en-US/docs/Web/API/Element/requestFullscreen

> ## Security
> Transient user activation is required. The user has to interact with the page or a UI element in order for this feature to work.

closed 2024-12-04 by andi34

florianibach, 2024-12-04:
@andi34 
I do not see this as a duplicate - since the feature requests, that the button should disappear. That is not what I need (I can live with the fullscreen button). 

What is happening here is, when I toggle the fullscreen button, take a photo (in fullscreen mode, which is working) and being on the summary page (also still in fullscreen) and then clicking on the "home" button on the top left, then the Fullscreen mode is being exited. 
Same behavior as when I click F5 in my browser. So it looks like with clicking the "Home" Button, the page is being refreshed

[image]

image summary in fullscreen mode

[image]
after clicking home, fullscreen is exited

andi34, 2024-12-04:
It is not possible, see:

> Transient user activation is required. The user has to interact with the page or a UI element in order for this feature to work.

andi34, 2024-12-04:
Besides that it's a duplicate. It was requested to have the fullscreen always on but it is technically impossible.

andi34, 2024-12-04:
I'll remove the option completely asap to avoid confusion. A relaunch of the page always exists the fullscreen mode and can't be triggered via scripts for security reasons. It's a browser thing independent of Photobooth.

florianibach, 2024-12-04:
@andi34 
I see these security reasons, I still did not get the reason why it is possible in Gallery and all the other back buttons to keep working and keeping in the fullscreen, while only the home button is not working 
(URL looks the same in both scenarios). 
But I do not understand many python code to be fair. 

I also want to say that I respect your descission to remove it - but maybe an alternative would be to add the description. Because maybe for some use cases (e.g. only displaying the gallery) it could be enough to let it run in fullscreen mode

Maybe for others who are coming across this bug: 
I tested the built in fullscreeen mode of the browsers in windows (press F11): this is working as intended, so even after taking the photo or refreshing the page, it is staying in fullscreen.

andi34, 2024-12-04:
> In a few words, the F11 fullscreen and fullscreen API are two different things. When pressing the F11 key the fullscreen feature is provided by the browser, not by the JavaScript API. This is a browser/OS default behavior.

Not sure if there's a real solution Photobooth side besides e.g. modifying Chromium source and compile chromium myself (won't do, would eat a lot of time without any benefits).

On windows and Linux Chrome, Chromium and Firefox can run in kiosk mode, android devices could use apps like fully kiosk browser.

Currently the option seems to only confuse user without real benefits. Photbooth reloads after some actions to reset some variables to it's defaults including its view.