PhotoboothProject/photobooth #1003: [BUG]: CUPS ratios options does nothing

theamnesic, 2024-12-05:
### Describe the bug

The printed image is slightly scaled (3mm), or the margins are negatives.
On the source jpg, the margins are the same as the middle ones (see attached photo).

### Steps to reproduce the behavior

Using a printer (DNP DS620) to print 4x6 paper, 300x300 dpi.

Tested with 1500x1000 and 1524x1016, 300 dpi collage.

The ratio options  'Shrink', 'Crop' or 'Expand' in CUPS does nothing.

### Expected behavior

Options to 'Shrink', 'Crop' or 'Expand' the collage in CUPS.

### Photobooth Version

v4.5.1

### Browser and browser version

Firefox up to date

### Operating System

Ubuntu 24.04

### Did it work before?

No

### Additional context

[Some](https://github.com/pibooth/pibooth/issues/152) [topics](https://sourceforge.net/p/gimp-print/support-requests/570/) talk about Gutenprint options. Maybe this option has to be passed by Photobooth to CUPS ?

[image]

andi34, 2024-12-05:
Please report to CUPS / Gutenprint devs, it's nothing Photobooth can change. Photobooth only uses the OS command line for printing.

andi34, 2024-12-05:
To use latest Gutenprint you can follow the tutorial/instructions here:
https://www.peachyphotos.com/blog/stories/building-modern-gutenprint/

Example usage of print command via Photbooth Adminpanel on how to pass supported options:

from a Mitsubishi 
```
lp -o orientation-requested=3 -o PageSize=w288h432-div2 -o fit-to-page -o StpColorPrecision=Best -o StpLaminate=Matte -o StpPrintSpeed=UltraFine %s
```

For Canon Selphy
```
lp -o media=Postcard.fullbleed -o landscape -o fit-to-page %s
```

andi34, 2024-12-05:
https://www.cups.org/doc/options.html

theamnesic, 2024-12-05:
Thank you @andi34!

I tried a looooooot of [options](https://wiki.debian.org/CUPSImageManipulation) (scaling, natural-scaling, margins, fill, etc. ) but it seem the printer force to fit (to much) the page.

Looks like it's not possible to have an image smaller than the paper size...

> Printer-specific options are also available and can be listed using the lpoptions command:

I tried to change the values `*ImageableArea w288h432/4x6: "0.000 9.120 297.600 451.680"` on the PPD file but nothing.

andi34, 2024-12-05:
Via command line you can define the options and use them independently of your settings via CUPS. There's nothing more you can do Photobooth side.

closed 2024-12-05 by andi34