PhotoboothProject/photobooth #1004: [BUG]: GPIO PINS not working on raspi5

mariusberlinghof, 2024-12-05:
### Describe the bug

im trying to make the button i'm using with my raspi5 photobox work.
I connected the button to NO 21 and Ground, the config in  `/boot/firmware/config.txt` is the following:
```
[all]
# Photobooth
#IN
gpio=5,6,7,8,16,17,20,21,22,26,27=pu
#OUT
gpio=9,10,11,12,18,19,23,24,25=op
# Photobooth End
```

I checked the button functionality with a python script:

``` python
import RPi.GPIO as GPIO
import time

# Define GPIO pin
PIN = 21

# Callback function for interrupt
def interrupt_callback(channel):
    print(f"Interrupt detected on pin {channel}!")

# GPIO setup
GPIO.setmode(GPIO.BCM)  # BCM pin numbering
GPIO.setup(PIN, GPIO.IN, pull_up_down=GPIO.PUD_UP)  # Enable pull-up resistor

try:
    # Add interrupt event detection
    GPIO.add_event_detect(PIN, GPIO.FALLING, callback=interrupt_callback, bouncetime=200)

    print("Waiting for interrupts. Press Ctrl+C to exit.")
    while True:
        time.sleep(1)

except KeyboardInterrupt:
    print("\nExiting...")

finally:
    GPIO.cleanup()  # Reset GPIO settings
```

Button support is enabled in settings and and the pin is set to 21.

Any idea on how to fix this / what i am missing?

### Steps to reproduce the behavior

1. install photobooth
2. connect hardware button to pin 21 andground
3. Check that  /boot/firmware/config.txt is set correctly (automatically done by setup script)
4. Try to take a photo with a button press (or collage with long button press)

### Expected behavior

A photo is taken and the on screen countdown timer starts

### Photobooth Version

4.99.0

### Browser and browser version

firefox 133.0

### Operating System

raspberry pi os with desktop

### Did it work before?

No

andi34, 2024-12-05:
Please read the FAQ. Pi foundations changed the sysfs handling kernel side. GPIO support only works on kernels below 6.6.

closed 2024-12-05 by andi34

andi34, 2024-12-05:
If you're screen is connected to the Pi you can use the key code feature and adjust the config.txt like this:
```
dtoverlay=gpio-key,gpio=21,active_low=1,gpio_pull=up,keycode=57
```
Keycode in adminpage `32` (for Space)

mariusberlinghof, 2024-12-06:
Thanks a lot for this tip, i missed out on the fact that im using kernel `6.6.51+rpt-rpi-2712`.
Thanks for the tip with the dt overlay, did i just miss this part in the faq?
Should i submit a change request for specifying the documentation and providing more details?

andi34, 2024-12-06:
The GPIO support will be removed if there's no other solution found besides downgrading the Kernel.

For the mentioned workaround you're welcome to add it to the documentation and submit the changes via pull request. Should fit best to the keycode trigger section as it's independent of the Remotebuzzer server and only works on local displays.

https://github.com/PhotoboothProject/photobooth/tree/dev/docs/faq

You could also run a local python script in background handling the button actions and using the http endpoints of the Remotebuzzer server - in case of a remote display (e.g tablet). This would also allow a wider range of Photobooth actions as keycode trigger only works for some actions.

andi34, 2024-12-06:
Most user already switched to remote trigger via ESP boards using the http endpoints of the Remotebuzzer Server.
Bluetooth also possible for keycode actions on ESP32.
https://github.com/PhotoboothProject/photobooth-ino