PhotoboothProject/photobooth #1005: [FEATURE]: Collage Preview

thetechnikfreak, 2024-12-05:
### Is your feature request related to a problem?

No

### Describe the solution you'd like

A Option to Show with part of the Collage you are actually taking at the time and a real time Preview where you can See the incomplete collagen coming togehte

### Describe alternatives you've considered

None

andi34, 2024-12-05:
Hey and thanks for your request. Sure we can think about a post processing after every taken image, but this would have an impact to the performance (especially to Raspberry Pi which are used a lot) and currently not sure if I'd like to go that step.
Also collage creation source is one of the parts need to be reworked further for next release.of Photobooth.

dave9123, 2026-01-16:
It could be disabled or enabled through the settings and have it automatically adjust for lower end processing power though :)