PhotoboothProject/photobooth #1071: [BUG]: Something went wrong please try again

JochenKlenk, 2025-02-11:
### Describe the bug

Hi,
I have a NikonCoolpix and Install on a Raspi 4 went well.
I can make pictures like oyu see here

[2025-02-11 22:23:03][main][DEBUG] capture.php []
[2025-02-11 22:23:03][main][DEBUG] Photobooth\PhotoboothCapture []
[2025-02-11 22:23:03][main][DEBUG] Capture with CMD {"cmd":"gphoto2 --capture-image-and-download --filename=%s","tmpFile":"/var/www/html/data/tmp/20250211_222303-0.jpg"}
[2025-02-11 22:23:42][main][DEBUG] returnData {"success":"collage","file":"20250211_222303.jpg","collage_file":"20250211_222303-0.jpg","current":0,"limit":4}
[2025-02-11 22:27:53][main][DEBUG] capture.php []
[2025-02-11 22:27:53][main][DEBUG] Photobooth\PhotoboothCapture []
[2025-02-11 22:27:53][main][DEBUG] Capture with CMD {"cmd":"gphoto2 --capture-image-and-download --filename=%s","tmpFile":"/var/www/html/data/tmp/20250211_222753.jpg"}
[2025-02-11 22:28:33][main][DEBUG] returnData {"success":"image","file":"20250211_222753.jpg"}

But then Error happens "Something went wrong....."

Let me know how I can send more details or maybe someone also have that issue.

Thanks

### Steps to reproduce the behavior

make a pictur eon website

### Expected behavior

Show taken picture

### Photobooth Version

newest

### Did it work before?

Yes

andi34, 2025-02-12:
Please set Loglevel 2 via Adminpanel and let me know the whole error message visible. I am not thinking there's a general bug, maybe configuration issue?

JochenKlenk, 2025-02-12:
Hi, the complete output of the debug log is shown above. Like I said the picture is taken and I can see the file on commandline. But only in tmp folder. not under images.

jochen@fotobox:/var/www/html/data/images $ ls
jochen@fotobox:/var/www/html/data/images $ cd ..
jochen@fotobox:/var/www/html/data $ cd tmp/
jochen@fotobox:/var/www/html/data/tmp $ ls
20250211_222103.jpg    20250211_222753.jpg  20250212_083315.jpg
20250211_222303-0.jpg  20250211_223026.jpg
jochen@fotobox:/var/www/html/data/tmp $

So maybe the issue is on transfering the file to the right folder

This is what I get now as error message

[image]

closed 2025-02-12 by andi34