PhotoboothProject/photobooth #1096: [FEATURE]: Support Autofocus for RPI Cam Module 3

szwenni, 2025-03-06:
### Is your feature request related to a problem?

Yes

### Description

When using rasperry pi with an attached camera module 3 (which has auto focus support), the integrated chromium browser is not able to use the autofocus as the autofocus functionality is integrated into libcamera.

### Describe the solution you'd like

I tried out different approaches including v4l2loopback, but the only one i got working was using gstreamer in combination with pipewire.
Packages needed for this are gstreamer, gstreamer1.0-libcamera, gstreamer1.0-pipewire
This command has to be executed before chrome/firefox is starting:
`gst-launch-1.0 libcamerasrc af-mode=continuous ! "video/x-raw,width=1280,height=720,format=YUY2",interlace-mode=progressive ! videoconvert ! pipewiresink mode=provide stream-properties="properties,media.class=Video/Source,media.role=Camera,node.nick=libcamera-cam"`

In chrome you have then a second camera called libcamera-cam, which will give you the camera output + autofocus.
I'm honestly not quite sure where to add this and how to set this up in the install script properly.
Maybe something like "Do you plan to use the Raspberry Pi with a screen and camera module 3 attached".

One thing I did not find out till now is how to disable the standard camera or how to give the libcamera-cam the standard. In chrome you can select which camera to use, maybe it makes sense when the camera is detected in javascript to look for the libcamera-cam device and take this instead of the first video device found.

### Describe alternatives you've considered

I tried to find out if somehow pipewire can natively advertise autofocus capability but this was a dead end somehow.
I tried with v4l2loopback but with this way chrome will not detect the cam.

andi34, 2025-03-06:
Afaik this is nothing Photobooth can change. It's an OS or Chrome related thing. Also see the discussions here https://github.com/RPi-Distro/chromium/issues/23#issuecomment-2426211732

Also autofocus is not an issue using go2rtc preview via URL.
https://photoboothproject.github.io/faq/#setting-up-a-preview-stream-from-your-dslr-or-picamera

closed 2025-03-06 by andi34

szwenni, 2025-03-06:
I agree with you partially here. Two things to add to this. Pipewire looks like the de facto standard now for wayland environments. Chrome and Firefox are using it by default now.

The other more intresting thing and I assume this is a RPI5 specific problem as it is missing a h264 hardware encoder.
Using go2rtc and rpicam-vid encoding the video to mjpeg will make the CPU have a default load of 60% as the encoding is full software. Additionally to the encoding there is the load of the streaming.

The proposed method also drives up the CPU load (also it might be different on a RPI4 with acceleration) but only to around 30-40%.

You are right that might be something which should be done by chrome or the OS. On the other side the kernel will most probably never support auto focus support through v4l2 as libcamera does this already in userland. And you are also right that it should be nativly supported by pipewire libcamera support to set the af mode, but as far as I looked through the pipewire support there is no such option.

You are right this is more of a workaround then a solution but for the RPI5 I found it at least valid to show as the CPU load really gave me headaches when I played around with encoding streams in earlier tries.