PhotoboothProject/photobooth #1097: [FEATURE]: GPIO handling for newer kernels

szwenni, 2025-03-06:
### Is your feature request related to a problem?

Yes

### Description

Right now it is not possible to use GPIO buttons with newer kernels as the sysfs support was dropped where the onoff nodejs library was relying on.

### Describe the solution you'd like

I'm currently working on a work in progress library which uses libgpiod to interface with GPIOs.
https://github.com/szwenni/libgpiod-node-wrapper

This issue should be used more or less as a discussion ticket.
The only point where I saw the usage of the GPIOs was here: https://github.com/PhotoboothProject/photobooth/blob/dev/assets/js/remotebuzzer-server.js

I'm aiming to provide a drop in replacement for onoff later on. For now I'm trying to find use cases what I additionally can test.
Currently I'm supporting reading, writing, watching, debouncing, info for lines, info for chips, listing chips and its labels and checking accessibility to the chips.
The problem is that I currently only have a RPI5 with kernel 6.6 to hand. I would like to get a few more tests before creating a pull request and publishing to npm.

### Describe alternatives you've considered

I checked onoff, I think this is not developed anymore since a few years.
I checked libgpiod-node, but this only supports an old version of libgpiod (<2.x), the newer version which is in dev has no watch support up to this time.

andi34, 2025-03-06:
I wasn't able to test, but I guess the already implemented way still works. Every gpio might need to be exported like this
```
echo 21 > /sys/class/gpio/export
```
To have the sysfs available.

szwenni, 2025-03-06:
Hi
Sadly is this not the case anymore. The GPIOs will have different numbers with different offsets e.g. GPIO 6 gets GPIO 518. In some cases the sys gpio layer does not even exist anymore as the distribution removed it and the plan is in future versions of the kernel to completly remove it.
I also read somewhere some days ago that the offsets cam change from boot to boot.

https://forums.raspberrypi.com/viewtopic.php?t=361116&start=125

This was the forum I orginally stumbled above. Especially the second post there.

Also just saw a hint in the faq for this problem:

https://photoboothproject.github.io/faq/#remotebuzzer-hardware-button-led-feature-using-gpio-connected-hardware-raspberry-pi-only

andi34, 2025-03-07:
When I looked last time the sysfs was enabled for all pi variants inside the PiOS kernel config. I am quite sure if needs the export to have the sysfs available on defined GPIO - might need the high value GPIO mapping. Most user would only have problems setting it up as it's complicated to setup via script automatically... which is why the note was added to the FAQ.